mjmeli/arduino-ad5933 #1: TODO: excitation voltage range, settling time cycles

mjmeli, 2017-04-09:
These need to be implemented. Anyone is welcome to contribute.

http://www.analog.com/media/en/technical-documentation/data-sheets/AD5933.pdf

Excitation voltage range: Table 10 (pg. 23)

Settling time cycles: Table 13 (pg. 25)

andrewbourhis, 2017-09-20:
Shouldn't it be a simple matter of the following:

`
#define CTRL_VOLT_RANGE1		(0b00000000)
#define CTRL_VOLT_RANGE2		(0b00000110)
#define CTRL_VOLT_RANGE3		(0b00000100)
#define CTRL_VOLT_RANGE4		(0b00000010)

/**
 * Set the excitation voltage range.
 *
 * @param range The voltage range
 * @return Success or failure
 */
bool AD5933::setExcitationVoltageRange(unsigned int range){
	byte val;
    if (!getByte(CTRL_REG1, &val))
        return false;

    // Clear out bits D9 and D10, which are the excitation voltage range bits
    val &= 0xF9;
    
	switch(range){
		case 1:
			val |= CTRL_VOLT_RANGE1;
		break;
		case 2:
			val |= CTRL_VOLT_RANGE2;
		break;
		case 3:
			val |= CTRL_VOLT_RANGE3;
		break;
		case 4:
			val |= CTRL_VOLT_RANGE4;
		break;
		default:
			return false;
	}
	return sendByte(CTRL_REG1,val);
}`

mjmeli, 2017-09-20:
Probably, this wasn't used in our implementation that used this library and so it was simply never written. If you have written this and it works then feel free to do a PR.

andrewbourhis, 2017-09-20:
Great, I'll be sure to do so when I am able to test it with hardware.

Thanks for posting this by the way, saved me some time!

qaisarabbasr, 2017-11-05:
has anyone built all the code involving CN-0217 and all the ranges. kindly attach code as it would be very helpful.I have made the hardware but finding difficulties in code.
Thanks
qaisar