mjmeli/arduino-ad5933 #2: Possible to program entire control register in one go?

andrewbourhis, 2017-09-20:
Wondering if there is something I am missing with regards to programming the control register. This implementation seems to separate each control register function which may be useful for readability, but requires multiple exchanges to be made between the AD5933 and uController. Wouldn't it be more efficient to set everything up first, then write a single time to the device control registers? I.e.:

` bool AD5933::setControlRegisters(byte reg1, byte reg2){
 	if(!sendByte(CTRL_REG1,reg1))
 		return false;
 	else
 		return sendByte(CTRL_REG2,reg2);
 }`

Where you first ensure reg1 and reg2 are properly set?

andrewbourhis, 2017-09-20:
Follow-up:

After looking more closely at your code, I see that you have a function for programming the mode - setControlMode(). However, I am confused on why you wipe the top 4 bits, since according to the datasheet for the AD5933, these bits control the mode, and bits D8 through D11 control excitation voltage and PGA gain.

mjmeli, 2017-09-20:
Yes this decision was originally made for readability and flexibility. i.e. we may not want to explicitly write all of the fields of the control register at once (i.e. allowing us to leave some as defaults) or we may wish to change a single function independently of having to know the state of the rest (i.e. we may want to change the clock source without having to know how to set everything else).

There's certainly nothing technical stopping a single write, it simply wasn't needed in the original implementation and so was never written, and you are welcome to add it.

Note that one overarching goal of the library was to prevent the person from using it to have to understand the bit manipulation required for the registers so preferably an implementation would accept arguments for each function (as the existing functions work) and then compile the byte to be sent itself rather than accepting the direct value.

andrewbourhis, 2017-09-20:
Gotcha, makes sense. Thanks for the speedy reply/explanation.