mjmeli/arduino-ad5933 #5: FAILED initialization

djbabai, 2018-01-11:
After uploading the code to an Arduino UNO, Started appears in the serial monitor shortly after which FAILED Initialization appears. I can't understand where the error is.

Monzurul-1, 2018-05-14:
I faced similar situation. It was solved when i put the start freq at 1000 or less !

AphroditeJoker, 2018-05-14:
It is because the Arduino Uno only has 2 byte for int, whose maximum is 32676. And that is approximately 2kHz I believe. If you do unsigned int I think it can goes up to 4kHz. If you wanna go higher than that you need a Arduino that has a SAMD chip on it(For example, Arduino Due). It will gives you 4 bytes int and you should be able to do 100kHz.

jhschwartz, 2019-04-09:
I think my pull request #11 might fix the problem here.

mjmeli, 2019-04-09:
Should be resolved by #11

closed 2019-04-09 by mjmeli