mjmeli/arduino-ad5933 #6: Any chance of an open-source license?

WolvesOfTheNight, 2018-03-30:
Any chance of posting a clear open-source license for this code? (Please point me to it if I somehow missed it.)

I know that the default of GetGub allows for forking projects, the GetHub ToS could be taken as an implicit license for any posted code, and that most people would rather not deal with this sort of thing. However, I would like to use a modified version of this code in my masters thesis. And, in short, if I modify it, I have to include it in the published thesis, and I have to receive copyright permission both to use it in the thesis, modify it, and to republish it. Worst case, I could just rewrite the thing, but I would rather not waste time rewriting it from the datasheet. An open-source licence tied to the project is an easy way to go, though I am open to discussing other options. Also, if you want a copy of the modified code, I am happy to send it over once it is finished. I will be publishing it, after all....

Thank you!

WolvesOfTheNight

mjmeli, 2018-03-30:
Hey, sure thing. I've added an MIT license. Feel free to submit a PR when done if you find any bugs or make any improvements.

closed 2018-03-30 by mjmeli

WolvesOfTheNight, 2018-03-30:
Thank you! I appreciate the prompt reply, and the clearly posted license meets school requirements. Once I am done I will see about a PR, and/or other distribution of the modifications, depending on the nature of the final version.