mjmeli/arduino-ad5933 #21: impedance measurement is not accurate

zifaum, 2021-12-08:
Hello sir
Thank you for your work sir, I have used your code sample sir.
I carried out impedance measurements with Rcal 10kohm and the results showed an accurate 10kohm, but when I took measurements on different resistors the results were far from accurate.
my example measured a 200kohm resistor but the results showed around 20kohm.
so how to solve this sir?

mkarn3yru96, 2022-04-14:
Hi @zifaum ! did you find a solution for this issue?

Hevaroo, 2022-08-29:
Hello @zifaum,

Depending on the setting you have used for the PGA-gain, excitation voltage range and value for Rfb, your limits of measurement (i.e min/max impedance) may vary. Could you please specify these parameters?