mjmeli/arduino-ad5933 #26: Has anyone implemented a phase measurement yet?

miemiemie666, 2023-03-15:
First of all, thank the author for sharing.
Has anyone figured out the phase? The true resistance and reactance values cannot be calculated without calculating the phase.
I tried several methods, but none of them worked out the phase.

miemiemie666, 2023-03-15:
The problem has been solved, just modify the library file as above, but int will cause errors, you need to further modify the int phase in the library to float phase type

eliecampa, 2023-12-07:
Dear all,
If someone is interested I wrote an example of using the AD5933 with a kind of command line using my own library (called Generic_MEAS) so you can easily add new command with int, float or long parameters. My example has a command called "FREQ, frequency (float)" so you can select the initial frequency and do the calibration (using some #define statement in the header file. After this FREQ command you can enter a ACQ command (without parameters) and got the results with phase correction based on calibration. The original has been modified an use a float as a frequency parameter and use mainly one frequency with a null frequency increment so the gain[I] is the mean gain and only valid for i=0. You can easily change the behavior of the software, for example the command FREQ can handle the start frequency, the increment frequency and the number of increments.
Let me know if you want to get the code.
Raymond

WheeSci, 2024-07-22:
To get around the problem with the phase being defined as an int, I added two new floats and converted the phase to degrees.
Also, why delete the arrays once you have them?  We want to use the phase when measuring unknown impedances.
code:
bool AD5933::calibrate(double gain[], int phase[], int ref, int n) {
    // We need arrays to hold the real and imaginary values temporarily
    int *real = new int[n];
    int *imag = new int[n];
	float faze, R;

    // Perform the frequency sweep
    if (!frequencySweep(real, imag, n)) {
        delete [] real;
        delete [] imag;
        return false;
    }

    // For each point in the sweep, calculate the gain factor and phase
    for (int i = 0; i < n; i++) {
        gain[i] = (double)(1.0/ref)/sqrt(pow(real[i], 2) + pow(imag[i], 2));
		R = float(imag[i])/real[i];
		faze = atan(R);     // page 19 of datasheet
		phase[i] = degrees(faze);
        // TODO: phase
    }

 //   delete [] real;
 //   delete [] imag;
    return true;